analogdevicesinc/arduino #13: Code wont compile to my arduino

louie132, 2019-09-05:
i am new to coding so don't know a lot but i followed the instructions from the website and i get a lot of errors for parts which aren't declared. I want to use a t type thermocouple with the EVAL CN0391 ARDZ on a arduino mega board can you give further guidance.

mirceacaprioru, 2019-09-05:
Hi,
The code is targeting the Arduino UNO board. I am not sure about the mega since it was not tested on that platform.

Could you give more details on what kind of errors you are getting ? 

Regards,
Mirea

louie132, 2019-09-05:
[image]

this is just part of the errors and is now being complied to an uno rev 3

thanks,
louie

louie132, 2019-09-05:
I copied and pasted all the libraries and code from the link but I don't know if i need to write extra information

mirceacaprioru, 2019-09-05:
You need to have all the files in that directory not just the *.ino file.

louie132, 2019-09-05:
which ones exactly i have all the ones in the CN0391_example folder

louie132, 2019-09-05:
[image]

mirceacaprioru, 2019-09-05:
copy them in the cn0391_example folder

louie132, 2019-09-05:
[image]

I then receive this problem

louie132, 2019-09-05:
[image]

sorry had moved the files instead of copied, now its copied i get a similar problem to the first

mirceacaprioru, 2019-09-05:
it seems there is an error with the includes.

there is
#include <file.h> replace it #include "file.h"

louie132, 2019-09-05:
Thank you this has eliminated a good majority of the initial problems.

[image]

any advice on this sorry there is so much