analogdevicesinc/arduino #14: Error in SPI interface for ADXL372

bbelson2, 2019-10-21:
In arduino/Arduino Uno R3/examples/ADXL372_example/adxl372.cpp 

The following utility function will fail for large FIFO requests:
```
static int adxl_read_reg_multiple(adxl_spi_handle *spi, unsigned char reg,
                                  unsigned char count, unsigned char *val)
{
    reg = reg << 1 | ADXL_SPI_RNW;
    return spi_write_then_read(spi, &reg, 1, val, count);   
}
```

The `unsigned char count` argument should become 16-bit (e.g. `unsigned short count`), since the FIFO buffer can contain up to 512 samples.

Best wishes,
Bruce

mirceacaprioru, 2019-10-22:
Hi Bruce,

You are right here. Thank you for catching this. Would you like to create a pull request with this fix ? 

Regards,
Mircea

mirceacaprioru, 2020-07-10:
Solved.

closed 2020-07-10 by mirceacaprioru

mirceacaprioru, 2020-07-10:
fixed with https://github.com/analogdevicesinc/arduino/pull/20