analogdevicesinc/arduino #19: Reading at max samplerate from ADXL372

Mopele, 2020-07-09:
Hey, I would like to get 3axis acceleration samples at atleast 1khz from the ADXL372. I would like to use a ESP32 for that. Do you think your libary is right for that and is this even possible. Where do I find an documentation about the wirering etc. Thanks a lot!

mirceacaprioru, 2020-07-10:
Hi,
You could use this code for the ADXL372, it is generic so you can use any platform you want for it:
https://github.com/analogdevicesinc/no-OS/tree/master/drivers/accel/adxl372

Some information about wiring can be found here : 
https://wiki.analog.com/resources/tools-software/uc-drivers/adxl372
https://wiki.analog.com/resources/eval/user-guides/arduino-uno/reference_designs/demo_adxl372
https://wiki.analog.com/resources/eval/user-guides/eval-adicup3029/hardware/adxl372
https://wiki.analog.com/resources/eval/user-guides/inertial-mems/accelerometers/adxl372

Regards,
Mircea

Mopele, 2020-07-13:
> Hi,
> You could use this code for the ADXL372, it is generic so you can use any platform you want for it:
> https://github.com/analogdevicesinc/no-OS/tree/master/drivers/accel/adxl372
>
> Some information [...]

Thanks for the great advice and the reference. But do you think it is possible to get the data with atleast 1k measurement points per second? I want to just print them out to the serial monitor via an microcontroller.
Regards!