analogdevicesinc/arduino #25: ADXL372 Not reading any data, ADXL_INT1_PIN never gets set to high

joshuaan7, 2021-03-25:
Hi, I'm using an EVAL-ADXL372-ARDZ shield along with Arduino Uno base board and Arduino IDE, and using the example program that has been provided to try and just read data and print it out on a serial monitor.

I can never read data because whenever I get to `digitalRead(ADXL_INT1_PIN)`, it reads as LOW which is not allowing me to move further along in the code and essentially not allowing me to read/print out any data. I've ensured that the `ADXL_INT1_PIN` pin is connected to pin 7 on the Arduino Uno, which is the default pin number. 

However, when I change the conditional statement from ` if (digitalRead(ADXL_INT1_PIN))` to ` if !(digitalRead(ADXL_INT1_PIN))`, (adding a NOT symbol in front of the conditional) I am able to get some output, although it is just zeroes. This makes me believe that there is something wrong with the pin connection on the hardware side of things.

Should I be doing something else to 'awaken' this pin or make this pin go HIGH? Or is there something wrong on the hardware end of things? Thank you for the help.  

```
void loop() {

  if (digitalRead(ADXL_INT1_PIN)) {
    delay(500);
    adxl372_Get_Highest_Peak_Accel_data(&adxl372, &accel_data);
  
    ...
  }
}
```

mirceacaprioru, 2021-03-26:
Hi,
Did you follow the instructions here:
https://wiki.analog.com/resources/eval/user-guides/arduino-uno/reference_designs/demo_adxl372

You may not have the CS connected correctly. 

Regards,
Mircea

joshuaan7, 2021-03-26:
> 
> 
> Hi,
> Did you follow the instructions here:
> https://wiki.analog.com/resources/eval/user-guides/arduino-uno/reference_designs/demo_adxl372
> 
> You may not have the CS connected correctly.
> 
> Regards,
> Mircea

Hi thanks for the quick reply,

I've followed the instructions in the link you've attached, and I can confirm that the CS_PIN = 10, which I believe is correct. 
I noticed that in those instructions, it says I need an Arduino Uno Rev 3. I am currently using an Arduino Uno R2. Could I still use this R2 board or do I need to use an Rev 3 board? 

Thank you,
Joshua

mirceacaprioru, 2021-03-26:
I think you are ok. After starting the project you should try creating am impact for the ADXL372 to sense it. What i do is try to impact it to the palm of my hand that should trigger the interrupt. I know it sounds silly but it is one way of generating high enough acceleration so the ADXL372 should detect it.

joshuaan7, 2021-03-26:
> I think you are ok. After starting the project you should try creating am impact for the ADXL372 to sense it. What i do is try to impact it to the palm of my hand that should trigger the interrupt. [...]

Okay I tried impacting it to the palm of my hand, and I get no readings, still not being able to trigger the interrupt to go HIGH. 
This is the only output I get when I tried impacting the sensor. 
`
Device id: 0
`

I also tried reading data when the interrupt remains LOW, but I continuously get 0.00 G data readings, which isn't helpful either.

mirceacaprioru, 2021-03-26:
Did you check the orientation of ADXL372 in the ARDZ shield ?

mirceacaprioru, 2021-03-26:
Since it is giving you Device id: 0 it means that the spi communication is not working. 

you should get Device id: 0xAD

joshuaan7, 2021-03-26:
> 
> 
> Since it is giving you Device id: 0 it means that the spi communication is not working.
> 
> you should get Device id: 0xAD

Oh okay that makes a lot of sense, do you know how I could fix this issue of spi communication not working?

mirceacaprioru, 2021-03-26:
The best way would be to use a logic analyzer to see if the commands are correctly transmitted to the device. 

But if you are using these standard adapters the ADXL372 with ARDZ shield just make sure the jumpers are in the correct position and the ADXL is connected in the ARDZ shield properly (check the pictures in the wiki page) 

A lot of times the SPI communication issue is related to  a pin mismatch between the SPI lines (MISO, MOSI, CLK, CS) on the Arduino and adxl372.

joshuaan7, 2021-03-29:
Closing this issue: found out it was an issue with the Arduino Uno board we were using. Once we replaced it with a different board, the program is working fine. Thanks for all the help.

closed 2021-03-29 by joshuaan7